adrianschlatter/threadlib #14: Collision Test

adrianschlatter, 2019-04-21:
Find a way to test for collisions between nut and corresponding bolt.

**Idea**

Render a nut and bolt combination (example: [tests/test.scad](https://github.com/adrianschlatter/threadlib/blob/master/tests/test.scad)). Use “number of volumes” that OpenSCAD shows after rendering:

- 3 volumes: PASS (1 volume nut, 1 volume bolt, 1 volume surroundings)
- 2 volumes: FAIL (collision)
- 1 volume: FAIL (something went terribly wrong)
- More than 3 volumes: FAIL (probably thread did not overlap bolt / nut and remained a separate part)

**Problem**

Find a way to make OpenSCAD tell us the number of volumes in a programmatic way (we don’t want to open OpenSCAD and hit F6 for hundreds of threads).

nh2, 2020-01-24:
You could use the `openscad` command line application (e.g. `openscad -o output.stl myfile.scad`), which outputs messages to `stderr`, and `grep` for the result.

adrianschlatter, 2020-01-24:
Unfortunately, it does not print the number of volumes. See discussion [here](http://forum.openscad.org/unit-testing-based-on-of-volumes-td26284.html).

steve8x8, 2024-01-04:
Since the forum has been closed and archived, here's the thread @adrianschlatter refers to: https://lists.openscad.org/empathy/thread/W3KYKIXDZDD2VQJUIZEKGBE6L3UTR5ZB