adrianschlatter/threadlib #17: Metric Trapezoidal Thread: TR

adrianschlatter, 2019-04-22:
Add TR-thread.

A good starting point (thread specs including tolerances) is [here.](https://mdmetric.com/tech/thddat16.htm).

adrianschlatter, 2020-06-20:
The link above does not work anymore.

fgoebel, 2021-09-10:
It is reachable per wayback machine: https://web.archive.org/web/20200311203631/http://mdmetric.com:80/tech/thddat16.htm

adrianschlatter, 2021-09-10:
Great that you are considering a contribution to threadlib 😀 

Yes, this part is a bit awk-ward. Too bad that OpenSCAD is no general purpose programming language. Otherwise I would have written everything as OpenSCAD code.

Where are you stumbling currently? What the awk-scripts are for? Or is it more how to write awk in general? Maybe I should add some documentation on how this works.

fgoebel, 2021-09-11:
I have only used awk for some simple tasks from cli, but never written any awk-scripts yet. But understanding the scripts is the easy part I think.

Your documentation is really good actually, but maybe you could make a sketch to explain where `Rrot` and `Dsup` are ending up in the final thread/bolt/nut?

As a first thing I added a table for TR Screws yesterday in #55.

I will try to calculate the required values for an example by hand and then use this to write the required awk script.
As far as i understand the threads are designed to be "in the between" the min and max values provided, for examlpe `d=(d_min+d_max)/2` ?

adrianschlatter, 2021-10-10:
Yes, threadlib aims for symmetric tolerances by aiming at the center of the range given in the norms. 

Regarding Rrot and Dsup (=2 * Rsup) I have extended the sketch in [Design of Threadlib](../blob/develop/docs/DesignOfThreadlib.md) to show where the support is and what Rrot and Rsup are supposed to be.