adrianschlatter/threadlib #26: SP-400 threads?

dholth, 2019-08-20:
Common American plastic & glass bottles seem to use these threads "SP-400": https://www.bevtech.org/threadspecs-downloads.asp

dholth, 2019-08-20:
See also http://www.gpi.org/industry-resources/finish-drawings

dholth, 2019-08-21:
`["PCF-33P-ext", [6.35, 14.74, 29.49, [[0, 0.76], [0, -0.76], [1.195, -0.549], [1.195, 0.07]]]],`

https://www.amazon.com/Motts-Apple-Juice-Pack/dp/B01IRHDZKY

adrianschlatter, 2019-10-10:
Sounds interesting. Thanks for this input. I'd like to add it but I would prefer to do it with some kind of test. Do you by any chance know the spec of the matching (internal) thread? I could then add both and the test would be to check for collisions between the two.

dholth, 2019-10-10:
No, as far as I know with bottle finishes the external thread on the bottle finish is standardized, but the caps are completely proprietary.

adrianschlatter, 2019-10-10:
Thanks for responding. This just tested that I properly receive notifications from Github (now ...).

Regarding SP400: That seems to be similar to PCO-1881 that also specifies the neck but not the cap. Maybe we could add a mating thread by hand and check for consistency. Did you produce a part with the [code you posted above](https://github.com/adrianschlatter/threadlib/issues/26#issuecomment-523619042)? Did it work?

dholth, 2019-10-10:
Yes I've printed lots of these now. They are working great.

adrianschlatter, 2019-10-10:
Aaah :) There is no better test than that! I'll gladly add it!

dholth, 2019-10-10:
Just make sure the cylinder supporting the threads is long enough that the bottle cap seals against the end of the tube, instead of ripping it off the object.

A nice feature of this kind of cap compared to the PCO-1881 apart from the larger size, is that the seal is flat instead of having to seal around a specific thickness of cylinder wall.

adrianschlatter, 2019-10-10:
One more question: I'm currently trying this and also checked [the spec](https://www.bevtech.org/assets/Threadspecs/pcf33p1.pdf). From the spec I interpret that the pitch should be 3.18 mm but your code suggests 6.35 mm. What am I getting wrong?

adrianschlatter, 2019-10-10:
Could you check whether PR #27 works for you?

dholth, 2019-10-10:
There's two thread starts! So you have to repeat the thread twice to make the screw. The distance between the two threads is half the distance between the same thread and itself.

adrianschlatter, 2019-10-10:
Got it. I‘ll have to think about that. I‘m not sure I‘m ready yet to extend threadlib to multiple thread starts.

dholth, 2019-10-10:
Something like this?
```
for (i=[0:starts])
   rotate([0,0,i*(360/starts)])
        thread(str(designator, "-ext"), turns=turns, higbee_arc=higbee_arc, fn=fn, table=table);
```

adrianschlatter, 2019-10-10:
Yes. But what I need to think about is
- does this belong in threadlib or is the user expected to write that loop? I think it belongs in threadlib.
- in THREAD_TABLE: Add new column "number of starts"?
- backwards compatibility: what will happen to those users having created their own private THREAD_TABLEs (without that "number of starts" column)? Will their code break? How many such users are there?
- What about unit testing?

dholth, 2019-10-11:
I applaud your effort to produce high quality threads.

alainchiasson, 2023-04-10:
Adding to the SP-400 thread thread. A few things for me have come together and I thought I would share and get feedback. I was looking to 3d print a neck my ketchup and shampoo bottles and stumbled upon the Glass Packaging Institute. First - the SP-400 designation seems to be a neck description: https://cnocoutdoors.com/blogs/blog/lets-talk-about-threads.

- sp-400 - 1 turn
- sp-410  - 1.5 turns
- sp 415 - 2 turns

dholth, 2023-04-10:
Yes, they say the caps are not standardized, only the bottle's neck finish.

alainchiasson, 2023-04-10:
I noticed that. I don't think that is "doable" in this lib beyond the thread, to me that's fine - someone else can do a "bottle neck" lib ;-)