adrianschlatter/threadlib #31: PCO-1881-int threads too thick

janrg, 2020-01-13:
The internal threads for PCO-1881 seem to be too thick.
https://i.imgur.com/SrbsaqO.jpg
It's a bit hard to see due to the material but compared to an actual PET bottle, the threads appear much thicker. A PET bottle lid will fit easily on an external thread made by threadlib but the lid I printed will lock up after less than half a turn on both the printed thread and a PET bottle.

adrianschlatter, 2020-01-20:
I'm not sure I understand what you mean. But there has been a bug with PCO-1881 internal thread that did not match an original PCO-1881 external thread (such as on a Coke bottle). It was close: You could make it go onto the external thread with a bit of force. Which version of threadlib did you use?

janrg, 2020-01-20:
Basically the lines of the internal thread are too thick (and thus the gaps between them too small). I used v0.3

adrianschlatter, 2020-01-21:
Ok. I understand that the problem is not with the diameters, but with the lengths in the direction of the axis. Is this collision occurring in the printed parts only, or already in the CAD?

janrg, 2020-01-22:
I didn't try for a collision in cad but comparing the 3d model with an actual bottle lid, the threads are *much* narrower there.

adrianschlatter, 2020-01-22:
I just verified the CAD in threadlib-0.3. Looks ok, no collision between bolt and nut. The PCO-1881-int thread may in fact be different from "real" parts because PCO-1881-int is - apparently - not standardised (only the external thread is). I assume this is a problem with tolerances: Not enough allowance in threadlib for your printing process to produce working parts.

If I created an internal thread with more allowance, would you be able to test it?

janrg, 2020-02-01:
Really sorry about the delay. Yeah I should be able to test it. I did manage to get things to fit by fiddling with the numbers in the thread table but since I don't entirely understand what the numbers mean, it's kind of haphazard :-)

adrianschlatter, 2020-02-03:
@janrg: I created an internal thread with grooves that are 0.2 mm wider than before. Can you test this commit? 5631169631bb3b730eb86ed52ad0fd2054cb1bec

BTW: What the numbers mean is explained [here](https://github.com/adrianschlatter/threadlib/blob/develop/docs/DesignOfThreadlib.md).

janrg, 2020-02-05:
Cool, will try this weekend.

janrg, 2020-02-08:
Works fine with an outer thread also generated in threadlib. Still binds on an actual PET bottle though. If you want to, I can fiddle with the numbers some more until it works with a PET bottle as well and then do a PR, but that might take a little while as my printer is quite busy at the moment :-)

adrianschlatter, 2020-02-09:
Thanks for testing, @janrg. It’s great that you help with improving threadlib! Apparently, the inner thread is printed with too much material (does not fit PET-bottle), while the external thread is printed with too little material (does fit printed external ). Is this normal? What kind of printer do you use? I have printed the (original) threadlib PCO-1881 using SLS and it did fit a PET-bottle.

janrg, 2020-02-09:
I use an FDM printer, Prusa i3 MK3S. I'm guessing that with SLS you get much tighter tolerances on threads since overhangs and the associated droop aren't a problem.

unrzn0, 2020-06-13:
Had very similar issues. The problem seems to be not threadlib itself but the 3D printers with inner threads. I‘ve created my own thread table with some extra tolerances to incresse the diameter of the thread slightly. This only changes the distance to the bolt and thus can be screwed nicely. Still to be used with care, especially if the thread should seal (use lots of PTFE tape then ...)

adrianschlatter, 2020-06-20:
I see. In that case, I assume that threadlib should remain as it is.

closed 2020-06-20 by adrianschlatter