adrianschlatter/threadlib #32: Threads Z-axis/height instead of turns

yhaenggi, 2020-01-31:
As a feature request, i'd like to create threads based on height and not amount on turns.

adrianschlatter, 2020-02-03:
Can you elaborate on how you would want to use that feature? What would the end-user code look like?

`thread("G1/2-ext", length=10);`?

yhaenggi, 2020-02-03:
Your example is exactly what would be needed. Currently i need to calculate or empirically find the amount of turns to get a nut/bolt/thread of length X.
A general length/height (however you want to name it) parameter would be suiting for this.
```
thread("G1/2-ext", length=10);
bolt("M3", length=10);
nut("M12x0.5", length=10, Douter=16);
```

unrzn0, 2020-04-11:
Hi, here's some modification for nuts and bolts using height.

```
// nut2 uses height instead of number of turns
module nut2(designator, height, Douter, higbee_arc=20, fn=120, table=THREAD_TABLE) {
    specs = thread_specs(str(designator, "-int"), table=table);
    P = specs[0]; Dsupport = specs[2];
    H = height;
    turns = (H / P) - 1;
    echo("Turns: ", turns);
    translate([0, 0, (P/2 + 0.1)]) {
    union() {
        thread(str(designator, "-int"), turns=turns, higbee_arc=higbee_arc, fn=fn, table=table);

        translate([0, 0, -P / 2])
            difference() {
                cylinder(h=H, d=Douter, $fn=fn);
                translate([0, 0, -0.1])
                    cylinder(h=H+0.2, d=Dsupport, $fn=fn);
            };
    };
    }
};
```

```
// bolt2 uses height instead of number of turns
module bolt2(designator, height, higbee_arc=20, fn=120, table=THREAD_TABLE) {
    specs = thread_specs(str(designator, "-ext"), table=table);
    P = specs[0]; Dsupport = specs[2];
    H = height;
    turns = (H / P) -1;
    echo("Turns: ", turns);
    translate([0,0,P/2]) {
    union() {
        thread(str(designator, "-ext"), turns=turns, higbee_arc=higbee_arc, fn=fn, table=table);
        translate([0, 0, -P / 2])
            cylinder(h=H, d=Dsupport, $fn=fn);
    };
    }
};
```

adrianschlatter, 2020-04-18:
Thanks for that example. What I do not like so much is the introduction of a new module. If someone wanted higbee_length instead of higbee_arc, we would already need nut3 and nut4. Adding inches instead of meters would already go up to nut8 (think of all the possible combinations). Nobody will be able to remember which module is for what :(

closed 2020-05-18 by adrianschlatter