adrianschlatter/threadlib #34: Thred for filter

kakl, 2020-04-10:
Hello Adrian, I am new in OpenScad. We are trying designing a reduction for particle filter against coronavirus. Can you, please, add a thread Rd 40x1/7" inner to your library? Thank you.

adrianschlatter, 2020-04-18:
Do you have a spec for Rd40x1/7“?

And out of curiosity: Do you have a link to your particle filter project?

kakl, 2020-04-18:
Hello, thank you for answer.
There are links to the [norm](https://grabcad.com/questions/i-can-not-model-the-screw-with-knuckle-threads-although-i-have-drawn-the-profile-i-am-attaching-the-files-please-how-can-create-the-rounded-thread-this-is-my-second-question)
And there are links to the projects:
[https://github.com/ODZ-UJF-AV-CR/COVID19SUBEA](https://github.com/ODZ-UJF-AV-CR/COVID19SUBEA) 
[https://drive.google.com/drive/folders/198LJC2vwE2VQVqj3EYvj3Ol7SrojzxHG](https://drive.google.com/drive/folders/198LJC2vwE2VQVqj3EYvj3Ol7SrojzxHG)

adrianschlatter, 2020-04-18:
Interesting project :)

Unfortunately, there are no allowance specs under that link and we can’t aim for 0 allowance because the printed threads would not fit due to production tolerances.

I would recommend that you define the threads in your private THREAD_TABLE using allowances that suit the precision of your 3D-printer (see here for explanation of how to create your own threads: [README](https://github.com/adrianschlatter/threadlib/blob/develop/README.rst)/extensibility and [Creating thread specs](https://github.com/adrianschlatter/threadlib/blob/develop/docs/CreatingThreadSpecs.md)).

closed 2020-05-18 by adrianschlatter