adrianschlatter/threadlib #37: problem with british pipe metrics

nohkumado, 2020-05-22:
Hello
i hade to make an adapter for a pump, and that pump uses instead of metric threading british non-SI metrics :( 
firing up your lib, i added a nut with  1/2" metric, and it exploded in my face....
doing just the thread was fine, but nut&bolt werent't ...
so i nagged the mailing list and Michael foudn out that there was no sanity check for those f...g imperial metrics, and that a fixed -int/-ext were added to the metrics....

i added those sanitize functions to make sure there is no double -int, or -int-ext on the name, and an assert to break the compilation with a n error message pointing to the source of the problem....
so, apart for my fervent, and unfortnately useless, wish that all those non-SI metrics could go down some very distant hell of their own, all my branch does is some parameter checking and correcting (if needed)...

adrianschlatter, 2020-05-22:
Can you post the code that „exploded in your face“?

nohkumado, 2020-06-09:
hello!
sorry for the delay, these mails get buried in the heap of system
messages...

here's the code, using your example on the page:

 10 //bolt("M4", turns = 5, higbee_arc = 30);
 11 bolt("G1/2-ext",turns = 5, higbee_arc = 30);

and that fails, wihtout letting to know why it failed, with my (and
Michaels) addition it now raises an alert about the name beeing not the one
expected/not in the table....

ciao
Bruno

adrianschlatter, 2020-06-09:
Ah. That is not related to imperial threads, then. The code fails due to the duplicate "-ext" (duplicate because a bolt always has external thread, it cannot have internal thread).

I understand that you wish threadlib to produce a more meaningful error message when someone tries to use a non-existing thread, right?

nohkumado, 2020-06-09:
exactly, with the stuff in my version of the tree an assert is issued when the lookup of the thread name fails (after trying to remove eventual duplicate -ext/-int from the name....

adrianschlatter, 2020-06-09:
Gaps in numbering?

adrianschlatter, 2020-06-09:
Oh, I just now understand that you apparently found that erroneous example somewhere in the threadlib docs. That would be a bug. Can you point me to the place where that misleading example is?

nohkumado, 2020-06-09:
Well its not really an erroneous example, its just misleading... you put an example of a metric nut, and after wards an example of juste the thread, perhaps a mention, careful, when using thread you aved to add -int/-ext to the name to have the 2 mesures of the thread....

well try for M45 .... doesn't exist... you have either M42 or M48... nothing in between...

adrianschlatter, 2020-06-09:
Ok. Regarding the misleading error message I'm all with you. So far, threadlib did not produce meaningful errors just because I have not been able to find a nice way to do it, as OpenSCAD has no exception handling and functions can have only a single expression. I have checked out your PR to find out how you did that.

BTW: There are 6 different types of M45 in threadlib :)

commit 703d6a88 (2020-06-09): thread_specs(...): Error Msg

Added helpful error message if thread_specs(...) is unable to
look up designator. Compare PR #36.

closes #37
closed 2020-06-09 by adrianschlatter