adrianschlatter/threadlib #38: tapered bolt?

nohkumado, 2020-06-09:
hello, i am trying to adapt your  lib to https://www.thingiverse.com/thing:3769527
there i want to have a tapered end, is this somehow possible?
some help/docu on how to create your own thread would be welcome too :D

adrianschlatter, 2020-06-09:
@nohkumado Tapered like a pipe thread (to prevent water from leaking)? That would not be high up on my priorities list but contributions are welcome (first step would be to obtain good specs). Looking at you part, I don't think this is what you want. Or do you mean tapered like in "lead-in / lead-out tapers"? If so, threadlib already does this for you.

nohkumado, 2020-06-09:
i meant like a wood screw, getting thinner to the point?

adrianschlatter, 2020-06-09:
Ok, that's (probably) something else again. I think in wood screws not only the radius of the helix changes but the profile of the thread, too. Unfortunately no: Nothing like that on the roadmap.

closed 2020-06-12 by adrianschlatter