adrianschlatter/threadlib #40: creating custom thread....

nohkumado, 2020-06-09:
Hello,
i am definitely no thread expert :( i tryed googling the terms used to specify the thread, but the only part that speaks to me is the pitch, the rest remains a vast mystery...

so i am quite stuck about how to fill in the data for my thread...
i tryed to copy over the data from a M48 thread, and tryed to fiddle around with it, but since each try uses up about 1.5min to compile, its hard to see what changes....
as soon as i change something randomly, the thread disappears... :( so having a plan on what to do would be nice...

i try to get a screw with an outer diameter of 50mm,  an inner diameter of 40, a pitch of 8mm, and will need the associated nut so what data changes/is impacted/how ?

thanks in advance!

nohkumado, 2020-06-09:
ok, finally understood some part of it... Rrot is the inner radius of the thread and just afterwards there is Dsupport as diameter of the inner circle of the thread.... and that leaves me puzzled? why the need to specify both? since one is 1/2 of the other??

nohkumado, 2020-06-09:
well....
now it displays in prerendering as it should but doesn't compile:
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330 

  aussenradius = 50;
 19           MY_THREAD_TABLE = [  
 20  ["M48p8-ext", [8, 19, 38,    [[0, -3.8], [0, 3.8], [5, 0.3], [5, -0.3]]]],
 21  ["M48p8-int", [8, -23, 45.0, [[0, 4], [0, -4], [4.9, -0.6], [4.9, 0.6]]]],
 22                    ];
 28          nut("M48p8", h = aussenradius/2, Douter = aussenradius,table=MY_THREAD_TABLE);

what's going wrong?

adrianschlatter, 2020-06-09:
I think you'll find the help you need in the docs: Read the markdown (.md files) in the docs subdir. The specification of the THREAD_TABLE contents is [here](https://github.com/adrianschlatter/threadlib/blob/develop/docs/DesignOfThreadlib.md).

nohkumado, 2020-06-09:
neat ev a pointer from the first page on it could be helpful?

still..... i don't get it to work for the -int....
 schraubend = 40;
    tol =0.5;

    MY_THREAD_TABLE = [ 
      ["M48p8-ext", [8, .99*(schraubend-tol)/2, schraubend-tol,    [[0, -3.8], [0, 3.8], [5, 0.3], [5, -0.3]]]],
      ["M48p8-int", [8, 1.02*(schraubend+tol)/2, (schraubend+tol), [[0, 4], [0, -4], [4.9, -0.6], [4.9, 0.6]]]],  
    ];
        specs = thread_specs("M48p8-int", table=MY_THREAD_TABLE);
        P = specs[0]; Dsupport = specs[2];
        H = (26) * P ;
        turns = (schraubend+tol)/P-1;
      thread("M48p8-int", turns =  turns, table=MY_THREAD_TABLE);
//difference()
//{
      cylinder(d=1.5*schraubend, h=25);
  //    translate([0,0,-1])cylinder(d=Dsupport, h=27);
//}     

as soon as i add the cylinder, it won't compile with the CGAL Error?
any idea?

adrianschlatter, 2020-06-20:
Looking at the internal thread spec, I believe there will be a problem because the thread will just touch itself after one full turn ([0, 4] to [0, -4] spans a distance of 8, the pitch is 8 as well). [0, 3.9], [0, -3.9] would avoid that problem.

closed 2020-07-09 by adrianschlatter