adrianschlatter/threadlib #47: WARNING: Can't open library 'scad-utils/lists.scad'.

dennisdebel, 2020-12-04:
As per title, there is no' lists.scad' in the latest scad-utils git either...

(OSX, 10.11, OpenScad 2019.05)

RajHid, 2020-12-11:
Had the same issue

Locked in the files and found some kind of advertising and Text [1] in thread_profile.scad
that dosen't belong in a *.scad file without out commenting marks.
--> how dos it get in the file? 
just delete these parts it and it worked for me.

I put the text that was the problem in these  <> code insertion markers. below
looks broken.
Same happens to the SCAD functionalety.

[1]
`Skip to content

    Why GitHub?
                          

                    
Team
Enterprise
Explore
                      

                    
Marketplace
Pricing
                       

                        

Sign in
Sign up
MisterHW /
IoP-satellite

0
10

    11

Code
Issues
Pull requests
Actions
Projects
Security

    Insights

Join GitHub today

GitHub is home to over 50 million developers working together to host and review code, manage projects, and build software together.
IoP-satellite/OpenSCAD bottle threads/thread_profile.scad
@MisterHW
MisterHW reorganize files into sub-folders
Latest commit 261b399 on Jul 15, 2018
History
1 contributor
134 lines (118 sloc) 4.35 KB
DELETE ABOVE =================================
 51  use <scad-utils/transformations.scad>    // https://github.com/openscad/scad-utils
...
181  [bottle_4841_neck_thread_height(),0.7],
182  [0,1.5]
183  ];
DELETE BELOW =================================
© 2020 GitHub, Inc.
Terms
Privacy

    Security
    Status
    Help
    Contact GitHub
    Pricing
    API
    Training
    Blog
    About`

RajHid, 2020-12-11:
Next problem is how do on get it working whit OpenSCAD-Nightly?
Problem is just to find the Libraries Folder.
I Found the recommended hints here [1]
tl:dr
libraries folder is in the home/USER/snap/openscad-nightly/2191/.local/share/OpenSCAD/libraries paht

[1]https://hackernoon.com/managing-ubuntu-snaps-the-stuff-no-one-tells-you-625dfbe4b26c

adrianschlatter, 2021-01-25:
@rajHid: Can you confirm that this problem is in thread_profile.scad and not in threadlib?

quirxi, 2021-03-04:
I got the same error message: "WARNING: Ignoring unknown module 'straight_thread', in file threadlib/threadlib.scad, line 31."
I fixed it somehow by putting all the necessary *.scad files from the included libraries and from the threadlib into one directory and adapting the "include" and "use" directive in that files.

OS: linux - debian version 10.8 (buster)
OpenSCAD: OpenSCAD-2021.01-x86_64.AppImage

adrianschlatter, 2021-03-04:
Hi @quirxi. Does your directory/file structure look as indicated in the [README.rst](https://github.com/adrianschlatter/threadlib/tree/master)?

quirxi, 2021-03-04:
Yes - I have tried the directory structure you suggested in the README and all sorts of other different directory structures, too. Not sure if i get this problem because I use the OpenSCAD appimage or because its a rather new version of OpenSCAD? But I can confirm that the libraries in the library folder under "/home/user/.config/OpenSCAD/libraries" have been found at least sometimes :P.

adrianschlatter, 2021-03-04:
Interesting. I just upgraded to OpenSCAD 2021.01 and had no problems rendering the examples in threadlib. I also checked whether thread_profile.scad had problematic updates but that does not seem to be the case. @RajHid had some funny looking text inside thread_profile.scad. Does this also happen to you, @quirxi?

quirxi, 2021-03-04:
The first time I tried to downlaod the thread_profile.scad file it was downloaded as a html file. When I realized it I used the "raw" button on github and download it as text. 
The artifacts @RajHid mentioned seem to be caused by the merge of two different versions of the same file in github!?
Maybe the cause is really the use of the Appimage version of OpenSCAD?

adrianschlatter, 2021-03-04:
I have no experience with the appimage. So yes, maybe it is related to this. Does it work for you when using the Debian package?

quirxi, 2021-03-04:
Now I have found the problem!
I have installed the libraries in "$HOME/.config/OpenSCAD/libraries" and *not* in "$HOME/.local/share/OpenSCAD/libraries" as described in "https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries".
Now the appimage and the default debian package both work as expected. 
I am really sorry for the work I have caused you, but I was so acustomed to the fact that local files are installed in "$HOME/.config/.." that I must have somehow igored the instructions. 
Thanks again for your efforts and your time!

adrianschlatter, 2021-03-05:
No problem, @quirxi . Thanks for sharing the solution :)

closed 2021-07-13 by adrianschlatter