adrianschlatter/threadlib #49: struggling to get a G1/2 nut fit on a G1/2 bolt

kermorgant, 2021-05-27:
Hi,

Thanks for this exciting lib !

I've 3d printed (with an ender 3) some bolts with [G1/2 threading using this lib](https://github.com/kermorgant/3dprojects/pull/3/files) and got good results, as I can fit standard gardening connectors.

Then I tried to print a nut as shown below that would fil but the result is way too tight.

```openscad
use <threadlib/threadlib.scad>

difference() {
  cylinder(h=5, d=28, $fn=6, center=true);
  translate([0,0,-3]) tap("G1/2", turns=6);
}
``` 

Given the bolt is printed like this, am I missing something or should the solution maybe reside in better printer settings ?

```openscad
use <threadlib/threadlib.scad>

turns = 7;
thread="G1/2-ext";

thread(thread, turns=turns);

specs = thread_specs(thread);

P = specs[0]; Rrot = specs[1]; Dsupport = specs[2];
section_profile = specs[3];
H = (turns + 1) * P;

difference() {
translate([0, 0, -P / 2])
    cylinder(h=H, d=Dsupport, $fn=120); /*  */
translate([0, 0, -P / 2 -1])
    cylinder(h=H+10, d=Dsupport-3, $fn=120); /*  */
}
```

adrianschlatter, 2021-05-28:
I modified your code a little so we can have a look at the fit:

```
use <threadlib/threadlib.scad>

turns = 7;
thread="G1/2-ext";

specs = thread_specs(thread);

P = specs[0]; Rrot = specs[1]; Dsupport = specs[2];
section_profile = specs[3];
H = (turns + 1) * P;

intersection() {
    union() {
        
        thread(thread, turns=turns);
        difference() {
        translate([0, 0, -P / 2])
            cylinder(h=H, d=Dsupport, $fn=120); /*  */
        translate([0, 0, -P / 2 -1])
            cylinder(h=H+10, d=Dsupport-3, $fn=120); /*  */
        }

        difference() {
          cylinder(h=5, d=28, $fn=6, center=true);
          translate([0,0,-3.63]) tap("G1/2", turns=6);
        }
    
    }
    cube(20);
}
```

This looks ok: I see no collisions between nut and bolt:

[image]

kermorgant, 2021-05-30:
Thanks, that was a nice way to "debug" the situation :smiley: 

I went on testing other print settings and with dynamic quality and the nut standing vertical, I got my first working result !

[image]

closed 2021-05-30 by kermorgant