adrianschlatter/threadlib #51: Explanation of [pitch, Rrotation, Dsupport, section_profile]

ameuret, 2021-07-06:
I need threads for the E26 lamp base which is widely used in 120V countries. I also need a 1/4 NPS thread. 

I'd be more than happy to put together PRs for them as soon as I _grok_ the specification variables used by `threadlib` and `thread_profile`.

Could someone who understands the meaning of these structures jot down a schematic on a paper napkin? ☺️

It would probably be helpful to many people willing to contribute more thread standards.

Still a noob on the subject but I've read many papers and it's a shame to waste time figuring out these parameters through trial and error.

adrianschlatter, 2021-07-10:
Hi @ameuret 

Contributions are always welcome. Regarding the specs used by threadlib: Does [this](https://github.com/adrianschlatter/threadlib/blob/develop/docs/DesignOfThreadlib.md) help?

ameuret, 2021-07-13:
That's awesome. It's awkward that I did not find it by myself... 😳 I'll submit something ASAP (currently away from home...)

nohkumado, 2021-10-04:
ok, i don't want to pollute with a separate thread, i am again at some custom thread  for a vacuum cleaner tube, and was pulling out my, inexistent, hair when i noticed that the thread is inversed from the one i was trying to implement, the docu is nice, but lacks that point, how do i inverse the rotation of the thread? i need one that goes from low right to left up?

adrianschlatter, 2021-10-06:
Hey nohkumado. Why not use the Discussion section for questions which are not related to issues in threadlib? You will even find an answer to you question there: [https://github.com/adrianschlatter/threadlib/discussions/53](https://github.com/adrianschlatter/threadlib/discussions/53)