adrianschlatter/threadlib #67: PCO-1881 for PET bottle threads not producing workable result

rlmalisz, 2022-02-05:
My code is crazy simple.  I have used your library for wine cap threads which worked great.  But every attempt I have made to print internal PCO-1881 threads has resulted in something that simply will not thread onto a bottle.  I am at version 0.3.  Code looks like:

nut("PCO-1881", turns=2, higbee_arc=45, Douter=32);

Printer is a Lulzbot Mini 2.  Have printed PCO-1881 bottle caps using Thingiverse STLs that seem decent.  Is there some trick to making this particular internal thread work?

adrianschlatter, 2022-02-10:
I did use a PCO-1881 internal thread and it did screw onto a PET bottle ([rocket fins](https://github.com/adrianschlatter/RocketFins)). Therefore, I'm rather confident threadlib is ok. Can you provide more detail on what is wrong with the thread you printed? Is the problem with diameters or is it maybe a pitch issue?

rlmalisz, 2022-03-13:
The "cap" using your threads neither starts well nor threads on straight...this is on an Aquafina 500ml bottle, as it's what I have to hand.  A cap with threads printed using threadlib crossthreads immediately.  Looking at the bottle's male threads, it looks like it has two starts, one 120 degrees clockwise from the other.  It also looks like the per-start "turn" on the bottle is about 0.5.  What values are you using that print working caps?

rlmalisz, 2022-03-13:
Disregard.  The damned Aquafina bottles appear to be PCF-26P thread.  On a real PCO-1881 bottle, your threads are a smidge snug (at least, the way they printed up for me using PETG) but are workable.  In a pinch, I can scale them out a fraction.