adrianschlatter/threadlib #80: M12 nut and M12 bolt do not fit

karlkashofer, 2023-05-26:
Hi ! Thanks for this fantastic and easy to use library ! I have just one problem, when i print a M12 nut and a M12 bolt they do not fit, the nut only screws on the bolt for one turn and then seizes up.
 I presume i need to add some more clearance ? How is that done properly ?

Thanks,
Cheers,
KK

Hiltronix, 2023-08-14:
Yes, I am having the exact same issue.  I used M60 as my experiment with this library.  And have the exact same result.  It seizes up after 1/2 turn.  

I've cleaned the threads, even removed material manually to make sure the threads are all clean enough.  Even shaved the threads down to allow tolerance.  But it makes no difference.  It's like the angles and spacing is all wrong, like when you try to mix incompatible thread types on metal nuts and bolts.

I've tried using the bolt function with nut, and bolt function with tap, and I get the exact same 1/2 turn seizing.

Could the author or maintainer please suggest why [karlkashofer] and myself are having these problems?

Thank you.

adrianschlatter, 2023-08-25:
Do the threads fit in CAD?

erydzewski55, 2023-09-05:
Having the same issue for M10 x 1

erydzewski55, 2023-09-05:
I am sorry, I do not know how to check that?

adrianschlatter, 2023-09-05:
You could create a nut and a bolt (or whatever you actually created), then "cut it in half" (both parts or just the nut) as in the picture below and then have a close look at it (particularly at the gaps between male and female threads).

[image]

erydzewski55, 2023-09-05:
From a static perspective look good, very tight, as mentioned they thread about a turn then stop. I have screw testers and the printed bolt fits the tester fine, but the nut does not. 

[image]
[image]

adrianschlatter, 2023-09-05:
Can you share the OpenSCAD code that generated the parts above?

erydzewski55, 2023-09-05:
I generated them separately by running the commands and exporting them to stl individually (super new to openScad so learning as I go lol)

bolt("M10x1", turns=6);
nut("M10x1", turns=10, Douter=12);

adrianschlatter, 2023-09-05:
Ok, I just ran this:
```
use <threadlib/threadlib.scad>

intersection() {
    union() {
            bolt("M10x1", turns=6);
            nut("M10x1", turns=10, Douter=12);
    };
    cube(100);
}
```

The results looks pretty ok to me:

[image]

erydzewski55, 2023-09-05:
I just tested with another nut I found elsewhere and the same issue occurs. I am going to try a different slicer.

erydzewski55, 2023-09-14:
After much investigation, it just seems to be a tolerance thing. The bolt prints and threads fine, but the nut does not. Scaling the nut up .4mm allowed for threading. I do not think there is any issue with the generation, just the the expansion when printing. thank you for the library and help trouble shooting!

closed 2023-10-18 by adrianschlatter