adrianschlatter/threadlib #85: recursion detected

rolfyone, 2024-01-07:
I need to say Im new to openscad...

i've setup the libraries as suggested:
```
➜  libraries ls
list-comprehension-demos thread_profile.scad
scad-utils               threadlib

```

then i try the sample script:
```
use <threadlib/threadlib.scad>
thread("G1/2-ext", turns=10);
```

I get:
```
Parsing design (AST generation)...
Saved backup file: /Users/paulharris/Documents/OpenSCAD/backups/unsaved-backup-xJL24641.scad
Compiling design (CSG Tree generation)...
[WARNING: Ignoring unknown function 'transform' in file ../thread_profile.scad, line 16](https://github.com/adrianschlatter/threadlib/issues/16,/Users/paulharris/Documents/OpenSCAD/libraries/thread_profile.scad) 
[WARNING: Ignoring unknown function 'transform' in file ../thread_profile.scad, line 16](https://github.com/adrianschlatter/threadlib/issues/16,/Users/paulharris/Documents/OpenSCAD/libraries/thread_profile.scad) 
[WARNING: Ignoring unknown function 'transform' in file ../thread_profile.scad, line 16](https://github.com/adrianschlatter/threadlib/issues/16,/Users/paulharris/Documents/OpenSCAD/libraries/thread_profile.scad) 
[WARNING: Ignoring unknown function 'transform' in file ../thread_profile.scad, line 16](https://github.com/adrianschlatter/threadlib/issues/16,/Users/paulharris/Documents/OpenSCAD/libraries/thread_profile.scad) 
[WARNING: Ignoring unknown function 'transform' in file ../thread_profile.scad, line 16](https://github.com/adrianschlatter/threadlib/issues/16,/Users/paulharris/Documents/OpenSCAD/libraries/thread_profile.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 62](https://github.com/adrianschlatter/threadlib/issues/62,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 62](https://github.com/adrianschlatter/threadlib/issues/62,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 62](https://github.com/adrianschlatter/threadlib/issues/62,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 62](https://github.com/adrianschlatter/threadlib/issues/62,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 62](https://github.com/adrianschlatter/threadlib/issues/62,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: Ignoring unknown function 'to_3d' in file skin.scad, line 11](https://github.com/adrianschlatter/threadlib/issues/11,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 37](https://github.com/adrianschlatter/threadlib/issues/37,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: undefined operation (undefined - undefined) in file skin.scad, line 37](https://github.com/adrianschlatter/threadlib/issues/37,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 67](https://github.com/adrianschlatter/threadlib/issues/67,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: undefined operation (undefined - number) in file skin.scad, line 67](https://github.com/adrianschlatter/threadlib/issues/67,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: len() parameter could not be converted in file skin.scad, line 37](https://github.com/adrianschlatter/threadlib/issues/37,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: undefined operation (undefined <= number) in file skin.scad, line 55](https://github.com/adrianschlatter/threadlib/issues/55,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[WARNING: undefined operation (undefined - number) in file skin.scad, line 56](https://github.com/adrianschlatter/threadlib/issues/56,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[ERROR: Recursion detected calling function 'distribute_extra_vertex' in file skin.scad, line 52](https://github.com/adrianschlatter/threadlib/issues/52,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[TRACE: called by 'distribute_extra_vertex' in file skin.scad, line 56](https://github.com/adrianschlatter/threadlib/issues/56,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[TRACE: called by 'insert_extra_vertices_0' in file skin.scad, line 37](https://github.com/adrianschlatter/threadlib/issues/37,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[TRACE: called by 'subdivide' in file skin.scad, line 37](https://github.com/adrianschlatter/threadlib/issues/37,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[TRACE: called by 'augment_profile' in file skin.scad, line 11](https://github.com/adrianschlatter/threadlib/issues/11,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos/skin.scad) 
[TRACE: called by 'skin' in file thread_profile.scad, line 26](https://github.com/adrianschlatter/threadlib/issues/26,/Users/paulharris/Documents/OpenSCAD/libraries/thread_profile.scad) 
[TRACE: called by 'straight_thread' in file threadlib.scad, line 31](https://github.com/adrianschlatter/threadlib/issues/31,/Users/paulharris/Documents/OpenSCAD/libraries/threadlib/threadlib.scad) 
[TRACE: called by 'thread' in file list-comprehension-demos, line 2](https://github.com/adrianschlatter/threadlib/issues/2,/Users/paulharris/Documents/OpenSCAD/libraries/list-comprehension-demos) 
Compiling design (CSG Products generation)...
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:02.517
```

I'm not sure what to do now, I can see the error, but its a different project, so would appreciate any guidance...

adrianschlatter, 2024-01-13:
Hi @rolfyone 

I just tried to run your code and it worked for me. Then, I upgraded my scad-utils (I was lagging a couple of commits behind the newest version), and it still works.

I use:

* OpenSCAD 2021.01
* scad-utils: newest (commit 26f8c2cf8124cdafe2fc21f9f9bbf2514e404834)
* list-comprehension-demos: newest (commit dec3fc3037f0107e5d9da79cb5b36b3e1ad9ce22)
* MisterHW/thread_profile.scad: newest (commit a78b36fc8fed4347eb05b7ef6d3df15df806ce55)

rolfyone, 2024-01-15:
Hrm im on master on https://github.com/OskarLinde/scad-utils/commit/31a4ab5b98e0a2e0531b13ee5c3a4ceb04137195

I can't see the scad-utils hash you're talking about, am i on the wrong library?

adrianschlatter, 2024-01-15:
I use openscad/scad-utils as linked in the Installation section of readme.md.

rolfyone, 2024-01-16:
> I use openscad/scad-utils as linked in the Installation section of readme.md.

ok that's brilliant, that's what i did wrong, i've used an scad-utils thats not the same! Thanks so much for your patience! closing :)

closed 2024-01-16 by rolfyone