adrianschlatter/threadlib #93: NPT threads

reinux, 2025-06-18:
Any chance we can get NPT threads? Thanks!

NulAsh, 2025-08-30:
I'm working on it:
```
use <thread_profile.scad>
eps=0.01;
$fn=360;
NPT2_profile=[
    [-0.0115996249670774927, -1.0430672674082538],
    [-0.07798914185021359, 1.0813972728521016],
    [1.67577597265425, 0.06886051189757472],
    [1.6800034811120326, -0.06641975875147245]
];
intersection() {
    cylinder(h=100, d=60.325);
    union() {
        cylinder(h=120, d1=55.8673, d2=55.8673+120*0.0625);
        translate([0,0,-2.208695652173913]) tapered_thread(section_profile = NPT2_profile, pitch = 2.208695652173913, turns = 40, r=55.8673/2-2.208695652173913*0.03125, higbee_arc=0, fn=360, left=false, taper=0.0625);
    }
}
```
This code uses `thread_profile.scad` from my fork of threadlib. This profile and pitch is for 1" to 2" NPT (11.5 TPI). Soon I will calculate for other sizes.

NulAsh, 2025-08-30:
Looks like I slightly f'd up the profiles - on the drawing in the standard the narrow end of the pipe is upside, so I calculated all the points with this assumption, but I'm starting with the narrow end downside. So the profile for TPI=11.5 should probably look like that:
```
NPT2_profile=[
    [-0.07798914185021359, -1.0813972728521016],
    [-0.0115996249670774927, 1.0430672674082538],
    [1.6800034811120326, 0.06641975875147245],
    [1.67577597265425, -0.06886051189757472]
];
```

NulAsh, 2025-08-30:
Yes, just turning it upside down should work. Here's the updated profiles:
For TPI=27:
```
profile=[
    [-0.03321759745472061, -0.46059513473330253],
    [-0.004940581004495969, 0.44426939167388586],
    [0.71555703825142132, 0.028289897245997524],
    [0.7137564327971805, -0.02932947728970775]
];
```
For TPI=18:
```
profile=[
    [-0.04982639618208091, -0.6908927020999538],
    [-0.0074108715067439537, 0.6664040875108288],
    [1.073335557377132, 0.042434845868996286],
    [1.0706346491957708, -0.043994215934561628]
];
```
For TPI=14:
```
profile=[
    [-0.06406250937696117, -0.8882906169856549],
    [-0.0095282633658136547, 0.8568052553710656],
    [1.38000285948488398, 0.05455908754585237],
    [1.3765302632517053, -0.05656399191586495]
];
```
For TPI=11.5:
```
profile=[
    [-0.077989141850213598, -1.0813972728521016],
    [-0.0115996249670774927, 1.0430672674082538],
    [1.68000348111203267, 0.06641975875147245],
    [1.67577597265425, -0.06886051189757472]
];
```
For  TPI=8:
```
profile=[
    [-0.112109391409682047, -1.554508579724896],
    [-0.016674460890173896, 1.4994091968993648],
    [2.415005004098547, 0.09547840320524164],
    [2.4089279606904843, -0.09898698585276366]
];
```
Everything else should be the same.

NulAsh, 2026-02-12:
@adrianschlatter , have you checked my code? I made a fork of `threadlib`, added `thread_profile.scad` to it with my modification to allow it to make conic threads.

adrianschlatter, 2026-02-12:
@NulAsh, I missed that. I see you found a quite elegant solution to add the conical tapering. I have no idea how conical threads are specified. Do you have links to (excerpts of) norms so I can have a look at that? I see you have already found a table with thread specs. Do you also have the pictures and explanations related to that?

adrianschlatter, 2026-02-14:
@NulAsh: That is a fantastic source :-) I guess the way to extend threadlib by conic threads, NPT in particular, would be approximately like this:

* obtain a license from @MisterHW to include thread_profile.scad in threadlib directly. Preferably he would commit it personally so that he visibly keeps the credit for his own work.
* include your tapered_thread() module (or merge it into the straight one with an added argument)
* go the standard route of [creating thread](https://github.com/adrianschlatter/threadlib/blob/develop/docs/CreatingThreadSpecs.md) specs for threadlib  (tabulate specs, write convertor to OpenSCAd, add unittest)